cogimon/rtt-gazebo-robot-sim #8: Gazebo-agnostic construction of a KinematicChain object

xwavex, 2016-06-01:
In order to use the kinematic chain representation in the [Proxy](https://github.com/cogimon/rtt-gazebo-robot-sim/blob/kinematic_chains/include/rtt-kinematic-chain-proxy.hpp) components that eventually create the interface which the users will see and which can unify different robots/chains in one kinematic chain, the [KinematicChain](https://github.com/cogimon/rtt-gazebo-robot-sim/blob/kinematic_chains/include/kinematic_chain.h) representation should not depend on the gazebo structured. Since these Proxy components don't have access to gazebo at all.
Currently it seems that everything is heavily tied to gazebo's model data type;
```c++
KinematicChain(const std::string& chain_name, RTT::DataFlowInterface& ports, gazebo::physics::ModelPtr model);
```

xwavex, 2016-06-01:
[image]

xwavex, 2016-06-01:
@EnricoMingo do you have some ideas to do this?

EnricoMingo, 2016-06-10:
My idea is that the HAL component will read the SRDF and generate the kinematic chain objects according to the parsing. Then the Kinematic Chain + Control Modes can take as well the SRDF ao maybe they can be configured by the user for particular purposes?

EnricoMingo, 2016-06-10:
The proxy component should be a sort of multyplexer of ports no?