cogimon/rtt-gazebo-robot-sim #14: Add interface to IMU and FT sensors

krprzemo, 2016-06-16:


EnricoMingo, 2016-10-17:
I will use this opened issue to describe the kind of data we need for both the sensors in simulation.

Staring from IMU, we reference to the file ImuSensor.hh in gazebo, the data simulated are:
- AngularVelocity: ignition::math::Vector3d, the angular velocity in the IMU sensor local frame
- LinearAcceleration: ignition::math::Vector3d, the imu linear acceleration in the IMU sensor local frame
- Orientation: ignition::math::Quaterniond, orientation of the IMU relative to a reference pose

For the Force/Torque sensor, we reference to the file ForceTorqueSensor.hh in gazebo, the data simulated are:
- Torque: ignition::math::Vector3d, the current joint torque
- Force: ignition::math::Vector3d, the current joint force
Note that the FT sensor is simulated as a joint.

ndehio, 2016-12-20:
I fully agree! As far as I remember, Joshua did something with a force/torque sensor plugin.

xwavex, 2017-05-18:
Force Torque Sensors are already integrated in the robot component and IMU is currently realized externally in the case of the COMAN robot

closed 2017-05-18 by xwavex