cogimon/rtt-gazebo-robot-sim #15: JointMapping Interface

xwavex, 2016-07-05:
Due to the obvious need of a _properly_ (cf. #13 ) organized joint mapping (cf. #5 ), we need to retrieve the information from **srdf/urdf** (cf. branch: https://github.com/cogimon/rtt-gazebo-robot-sim/tree/URDF_SRDF_load) and provide it to other Orocos RTT components. For the second part however, we need a specified interface. I suggest we utilize the build-in mechanism for defining an interface in Orocos RTT.

EnricoMingo, 2016-07-05:
There is a branch with everything already implemented, may you take a look?

xwavex, 2016-07-05:
@EnricoMingo do you mean this branch: https://github.com/cogimon/rtt-gazebo-robot-sim/tree/temporary_implementation ?

EnricoMingo, 2016-07-05:
No, URDF_SRDF_load

xwavex, 2016-07-05:
Ok, sure I know that. That's why I referenced that branch in the description of this issue. However, I am more concerned about the interface to provide the mapping to other components. Since the robot-sim already loads everything from the srdf/urdf, we can get the information from that component, instead of having every component use the parser and parse the urdf/srdf by itself.

Right now we have to functions inside:
```c++
std::map<std::string, std::vector<std::string> > getKinematiChainsAndJoints();
```
and
```c++
/**
 * Provides the joint name to index mapping for other components to retrieve.
 * If there isn't such an port (portName) existing, or used in an kinematic chain,
 * the call will return an empty map. Otherwise it will contain the mapping.
 */
std::map<std::string, int> getJointMappingForPort(std::string portName);
```

If I remember correctly last time we agreed on having a `vector<std::string>` for each kinematic chain that contains the different joints in the correct order depending on their indices.

So, what I am talking about is how to define the correct interface specification and implement it according to the Orocos' built-in interface mechanisms.

EnricoMingo, 2016-07-05:
I have done the operations:
std::map<std::string, std::vector<std::string> > getKinematiChainsAndJoints();
to retrieve all the informations regarding number of kinemtac chains, names and joints (with the right order).

xwavex, 2016-07-05:
Ok, then we'll take this as base and need to "wrap" it in the Orocos interface mechanism. Also we do need to have a constraint that ensures that this operation can only to be called once the urdf/srdf is parsed. On the other side, the components that retrieves the joint mapping through this operation, should only be able to be configured once they retrieved the mapping.

EnricoMingo, 2016-07-05:
there should be something similar inside...what does it means to "wrap" it in the Orocos interface mechanism?

xwavex, 2018-01-11:
@EnricoMingo do we still care about this issue? Can it be closed?

EnricoMingo, 2018-01-11:
You can close

closed 2018-01-11 by xwavex