cogimon/rtt-gazebo-robot-sim #18: Can not import it in Deployer

sinamr66, 2016-09-21:
Apparently, some is wrong with the version 327a65373e02919b1ada3f0f8041a522d4d46eb8 as we can not import "rtt-gazebo-robot-sim" in the deployer anymore ....The error is as follow:
Deployer [S]> import("rtt-gazebo-robot-sim") 
20.002 [ ERROR  ][ComponentLoader::import(path_list)] Unloading /vol/cogimon/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: not a valid component library:
20.002 [ ERROR  ][ComponentLoader::import(path_list)]    /vol/cogimon/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined symbol: createComponent
20.002 [ ERROR  ][ComponentLoader::import(path_list)]    /vol/cogimon/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined symbol: getComponentType
20.003 [ ERROR  ][Logger] Exception raised while executing an operation : Some found plugins could not be loaded !
20.003 [ ERROR  ][Logger] in Deployer: unhandled exception in sent operation.
The command 'import("rtt-gazebo-robot-sim")' caused a std::exception: 'Unable to complete the operation call. The called operation has thrown an exception' and could not be completed.

I've tested this version "b34eae3bbb7fd46b9c7e715d348b28f07a32ba11" can it works fine.
Any suggestion?

EnricoMingo, 2016-09-21:
Hello, you should update also cogimon-gazebo-models repo and if you update cogimon-scenarios you will find the new .ops files for import the last version of the component.

sinamr66, 2016-09-21:
Thx  alot for a quick reply. 
Actually, everything is updated....I just installed Jenkins on a fresh Ubuntu installed PC and then I ran this example : "coman_bring_up_kinchains.ops" from tutorial...

EnricoMingo, 2016-09-21:
Ok, try coman_bring_up.ops .

sinamr66, 2016-09-21:
Same Error..:(

sinamr66, 2016-09-21:
62.005 [ ERROR  ][ComponentLoader::import(path_list)] Unloading /vol/cogimon/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: not a valid component library:
62.005 [ ERROR  ][ComponentLoader::import(path_list)]    /vol/cogimon/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined symbol: createComponent
62.005 [ ERROR  ][ComponentLoader::import(path_list)]    /vol/cogimon/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined symbol: getComponentType
62.005 [ ERROR  ][Logger] Exception raised while executing an operation : Some found plugins could not be loaded !
62.005 [ ERROR  ][Logger] in Deployer: unhandled exception in sent operation.
The command 'import("rtt-gazebo-robot-sim")' caused a std::exception: 'Unable to complete the operation call. The called operation has thrown an exception' and could not be completed.

EnricoMingo, 2016-09-21:
I suppose the path inside the ops files are wrong, you should change them
for your machine

xwavex, 2016-09-21:
@sinamr66 in the Jenkins, did everything build successfully? Especially the rtt-gazebo-robot-sim? For me it seems like there is an error related to the **libgain_parser-gnulinux.so** library.

EnricoMingo, 2016-09-21:
Dennis, may you check the  CMake file of rtt-gazebo-robot-sim?

sinamr66, 2016-09-22:
@EnricoMingo I double checked the path, and I tried on another machine to make sure, the path is right :( 
@xwavex Yes. all of them are sunny... :(...I think there is a problem with the linked libraries, for createComponent and getComponentType functions....I've tied to find the source..But I failed:(

xwavex, 2016-09-22:
I just checked the following commit: https://github.com/cogimon/rtt-gazebo-robot-sim/commit/a518e5e8d413e0ee3ddfd5af35337778117b8116

What was the reason for changing **-ADD_LIBRARY(gain_parser  src/parser.cpp)** to **+orocos_component(gain_parser  src/parser.cpp ${rtt_gazebo_robot_sim_INCLUDES})** ?
I assume the gain_parser is a pure library and not an orocos component, isn't it? So ADD_LIBRARY should be used I guess.
By the way, @EnricoMingo is the current version working for you with this cmake currently?

EnricoMingo, 2016-09-22:
> I just checked the following commit: a518e5e

Yes...after lunch I will cancel and tebuild everything from scratch and let
u know

EnricoMingo, 2016-09-22:
I have just erased the whole build folder of rtt-gazebo-robot-sim and recompiled. 
This is the output if I run `deployer-gnulinux coman_bring_up.ops`:

>Real-time` memory: 517904 bytes free of 524288 allocated.
>Gazebo multi-robot simulator, version 6.6.0
>Copyright (C) 2012-2015 Open Source Robotics Foundation.
>Released under the Apache 2 License.
>http://gazebosim.org

 >8.171 [ Warning][Thread] Forcing priority (10) of thread with SCHED_OTHER policy to 0.
>[Msg] Waiting for master.
>[Msg] Connected to gazebo master @ http://127.0.0.1:11345
>[Msg] Publicized address: 10.255.32.75
>[[--- Gazebo running ---]]
>10.763 [ ERROR  ][ComponentLoader::import(path_list)] Unloading /home/enrico/robotology->superbuild/build/install/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: not a valid >component library:
>10.763 [ ERROR  ][ComponentLoader::import(path_list)]    /home/enrico/robotology->superbuild/build/install/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined >symbol: createComponent
>10.763 [ ERROR  ][ComponentLoader::import(path_list)]    /home/enrico/robotology->superbuild/build/install/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined >symbol: getComponentType
>11.460 [ ERROR  ][ScriptingService] Exception raised while executing an operation : Some found >plugins could not be loaded !
>11.461 [ ERROR  ][ScriptingService] in Deployer: unhandled exception in sent operation.
>11.462 [ Warning][Thread] Forcing priority (11) of thread with SCHED_OTHER policy to 0.
>Gazebo multi-robot simulator, version 6.6.0
>Copyright (C) 2012-2015 Open Source Robotics Foundation.
>Released under the Apache 2 License.
>http://gazebosim.org

>11.538 [ Warning][ScriptingService] Done configuring component
   >Switched to : Deployer

 >This console reader allows you to browse and manipulate TaskContexts.
  >You can type in an operation, expression, create or change variables.
  >(type 'help' for instructions and 'ls' for context info)

  > TAB completion and HISTORY is available ('bash' like)

  >  Use 'Ctrl-D' or type 'quit' to exit this program.

>Deployer [X]> ls

> Listing TaskContext Deployer[X] :

 >Configuration Properties: 
  >   string RTT_COMPONENT_PATH = /home/enrico/robotology->superbuild/orocos_ws/install_isolated/lib/orocos:/opt/ros/indigo/lib/orocos:/home/enrico/robotology-superbuild/build/install/lib/orocos:/opt/ros/indigo/lib/orocos/gnulinux/plugins:/opt/ros/indigo/lib/orocos/gnulinux/types:/home/enrico/robotology-superbuild/build/external/rtt-gazebo-lwr4plus-sim/orocos: (Locations to look for components. Use a colon or semi-colon separated list of paths. Defaults to the >environment variable with the same name.)
  >     bool AutoUnload     = true                 (Stop, cleanup and unload all components loaded by the >DeploymentComponent when it is destroyed.)

> Provided Interface:
 > Attributes   : 
  >   string Target         = gnulinux            
    >   bool Valid          = false               
   >     int ORO_SCHED_RT   = 1                   
    >    int ORO_SCHED_OTHER = 0                   
   >     int LowestPriority = 1                   
  >      int HighestPriority = 99                  

 > Operations      : activate addPeer aliasPeer cleanup cleanupComponents clearConfiguration configure >configureComponents connect connectOperations connectPeers connectPorts connectServices >connectTwoPorts createStream displayComponentTypes error getComponentTypes getCpuAffinity >getPeriod import inFatalError inRunTimeError isActive isConfigured isRunning kickOut kickOutAll >kickOutComponent kickStart loadComponent loadComponents loadConfiguration >loadConfigurationString loadLibrary loadService path reloadLibrary removePeer runScript setActivity >setActivityOnCPU setCpuAffinity setFileDescriptorActivity setMasterSlaveActivity setPeriod >setPeriodicActivity setSequentialActivity setSlaveActivity start startComponents stop stopComponents >stream trigger unloadComponent unloadComponents update waitForInterrupt waitForSignal 

 >Data Flow Ports: (none)

 >Services: 
 >      scripting ( Orocos Scripting service. Use this service in order to load or query programs or state >machines. ) 

> Requires Operations :  (none)
> Requests Services   :  (none)

> Peers        : coman[S] gazebo[R] 

Despite the errors the simulation properly starts...

kevinkim84, 2016-09-23:
Hi guys, 
Yes, we can start the simulation even with the errors, but following is what we get when we run coman_bring_up.ops :

-----------------------------------------------------------------

30.116 [ ERROR  ][ComponentLoader::import(path_list)] Unloading /home/parallels/citk/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: not a valid component library:
30.116 [ ERROR  ][ComponentLoader::import(path_list)]    /home/parallels/citk/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined symbol: createComponent
30.116 [ ERROR  ][ComponentLoader::import(path_list)]    /home/parallels/citk/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined symbol: getComponentType
30.122 [ ERROR  ][ScriptingService] Exception raised while executing an operation : Some found plugins could not be loaded !
30.123 [ ERROR  ][ScriptingService] in Deployer: unhandled exception in sent operation.
30.123 [ Warning][Thread] Forcing priority (11) of thread with SCHED_OTHER policy to 0.
30.126 [ ERROR  ][ScriptingService] Exception raised while executing an operation : Could not open file /robots/cogimon-gazebo-models/iit-coman/model.urdf for parsing.
30.126 [ ERROR  ][ScriptingService] in coman: unhandled exception in sent operation.
Gazebo multi-robot simulator, version 6.6.0
Copyright (C) 2012-2015 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

30.127 [ ERROR  ][ScriptingService] left_arm is not available!
30.127 [ ERROR  ][ScriptingService] right_arm is not available!
30.128 [ ERROR  ][ScriptingService] No such port: 'torso_JointPositionCtrl' while looking for port 'coman.torso_JointPositionCtrl'
30.128 [ ERROR  ][ScriptingService] No such port: 'left_arm_JointPositionCtrl' while looking for port 'coman.left_arm_JointPositionCtrl'
30.128 [ ERROR  ][ScriptingService] No such port: 'right_arm_JointPositionCtrl' while looking for port 'coman.right_arm_JointPositionCtrl'
30.129 [ ERROR  ][ScriptingService] No such port: 'left_leg_JointPositionCtrl' while looking for port 'coman.left_leg_JointPositionCtrl'
30.129 [ ERROR  ][ScriptingService] No such port: 'right_leg_JointPositionCtrl' while looking for port 'coman.right_leg_JointPositionCtrl'
 = true   

-----------------------------------------------------------------

Sina and I are both having the same problem. We recently built the whole thing on new machines through Jenkins so both cogimon-gazebo-models and cogimon-scenarios should be up to date.

EnricoMingo, 2016-09-23:
Ok so the plugin is loaded anyway but we have the error. I think is
something related with the CMakes and paths.

xwavex, 2016-09-23:
Hi I think everyone has the error and I noticed that it really is related to the following statement in the cmake:

```
10.763 [ ERROR ][ComponentLoader::import(path_list)] Unloading /home/enrico/robotology->superbuild/build/install/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: not a valid >component library:
10.763 [ ERROR ][ComponentLoader::import(path_list)] /home/enrico/robotology->superbuild/build/install/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined >symbol: createComponent
10.763 [ ERROR ][ComponentLoader::import(path_list)] /home/enrico/robotology->superbuild/build/install/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser-gnulinux.so: undefined >symbol: getComponentType
```

As I mentioned earlier, we should use
```
ADD_LIBRARY(gain_parser src/parser.cpp)
```
instead of
```
orocos_component(gain_parser src/parser.cpp ${rtt_gazebo_robot_sim_INCLUDES})
```
in the cmake...

xwavex, 2016-09-23:
@kevinkim84 Looking at the log you've posted, I guess your problem is in particular the path to the urdf:  ```
Exception raised while executing an operation : Could not open file /robots/cogimon-gazebo-models/iit-coman/model.urdf for parsing.
```
for me it seems like there is something missing. The path looks broken...

EnricoMingo, 2016-09-23:
Yes indeed, that one is the path of the coman-gazebo-models in the iit superbuild

xwavex, 2016-09-23:
Ah ok good to know

xwavex, 2016-09-23:
@kevinkim84 Please check if the following folder is existing for you:
```
/home/parallels/citk/cogimon-minimal-nightly/share/gazebo/models/cogimon/iit-coman
```
If so, then you have to edit the **coman_bring_up.ops**:
Change the paths in these line from:
```
var string urdf_path = os.getenv("ROBOTOLOGY_ROOT") + "/robots/cogimon-gazebo-models/iit-coman/model.urdf"
var string srdf_path = os.getenv("ROBOTOLOGY_ROOT") + "/robots/cogimon-gazebo-models/iit-coman/coman.srdf"
```
...to:
```
var string urdf_path = "/home/parallels/citk/cogimon-minimal-nightly/share/gazebo/models/cogimon/iit-coman/model.urdf"
var string srdf_path = "/home/parallels/citk/cogimon-minimal-nightly/share/gazebo/models/cogimon/iit-coman/coman.srdf"
```

xwavex, 2016-09-23:
We are currently working on a solution that works for everyone tho, since we hate hard-coded paths.

EnricoMingo, 2016-09-23:
I agree

xwavex, 2016-09-23:
I fixed the cmake issue:

1. you need to make sure to delete this file **/home/parallels/citk/cogimon-minimal-nightly/lib/orocos/gnulinux/rtt-gazebo-robot-sim/libgain_parser.so** if it exists for you.
2. rebuild the **rtt-gazebo-robot-sim** with Jenkins again

xwavex, 2016-09-23:
FYI the error was triggered because Orocos tried to load the libgain_parser.so as Orocos Component LIbrary, just because it was installed in the folder for component libraries. This was destined to fail. because a normal shared library does not have the specific entry points required by Orocos. So I changed the install path to the normal lib folder.

sinamr66, 2016-09-23:
I missed a lot of replies..:D Thx a lot for the helps..:)
@xwavex  Thx a lot..:)...We can import("rtt-gazebo-robot-sim") it now..:)
But we can not communicate with the robot yet, even-though we did these changes:
`var string urdf_path = "/home/parallels/citk/cogimon-minimal-nightly/share/gazebo/models/cogimon/iit-coman/model.urdf"`
`var string srdf_path = "/home/parallels/citk/cogimon-minimal-nightly/share/gazebo/models/cogimon/iit-coman/coman.srdf"`

We get the following errors:
`Deployer [S]> coman.loadURDFAndSRDF(urdf_path, srdf_path)
ERROR: get_joints_in_chain() failed
ERROR: requested chain in get_enabled_joints_in_chain() does not exist.
ERROR: requested chain in get_enabled_joints_in_chain() does not exist.
ERROR: requested chain in get_enabled_joints_in_chain() does not exist.
ERROR: requested chain in get_enabled_joints_in_chain() does not exist.
 = false  `
The error we got are :

sinamr66, 2016-09-23:
It worth to say that the problem of the path is solved....This error is new..:(

xwavex, 2016-09-28:
> Did you download https://github.com/corlab/xbotcoremodel ?

yeah he did, if it is related to that then i might have the same problem,
no?

2016, 10:18:

sinamr66, 2016-09-28:
@EnricoMingo  is it a part of Jenkins? If yes, yes we did..:)

xwavex, 2016-12-14:
I think this is fixed after the workshop, isn't it?!

closed 2016-12-14 by xwavex

sinamr66, 2016-12-14:
Yes. Thx ..:)