cogimon/rtt-gazebo-robot-sim #19: Simulation of LoadCells

EnricoMingo, 2016-10-17:
We will need also Load Cells (FSR) sensors simulation since they will be mounted down the feet. 
In NAO these sensors are implemented as bumper sensors using the `libgazebo_ros_bumper.so` (check [https://github.com/ros-naoqi/nao_robot/blob/master/nao_description/urdf/naoV33_generated_urdf/naoGazebo.xacro](url)

The `libgazebo_ros_bumper.so` ([https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_bumper.cpp](url)) is a particular implementation of the contact sensor in gazebo ([https://bitbucket.org/osrf/gazebo/src/502ac23ec4388fdf1e004dc09dabbb86442bd5e5/gazebo/sensors/ContactSensor.hh?at=default&fileviewer=file-view-default](url)).

Here the information is at least the tangential force sensed by the sensor (that should be just a double) in the frame of the sensor.

EnricoMingo, 2016-10-17:
Note that considering the documentation of the contact sensor [https://bitbucket.org/osrf/gazebo/src/502ac23ec4388fdf1e004dc09dabbb86442bd5e5/gazebo/sensors/ContactSensor.hh?at=default&fileviewer=file-view-default](url), we need to rotate the force reading from inertial frame to local (sensor) frame. If we are going to develop a brand new plugin I suggest to be compliant with the ROS one.

xwavex, 2016-11-04:
What does "frame" mean in these cases? Is it a reference to a pose (= translation and rotation) or is it a pose itself?

xwavex, 2016-11-04:
BTW since this is a very general question for sensor support, perhaps we should move it to another repo.

EnricoMingo, 2016-11-04:
The frame associated to the sensor.

xwavex, 2016-11-04:
Yes but how is it presented concretely?

EnricoMingo, 2016-11-04:
Normally depends on the particular sensor plugin (most of the times you can
set it basically from the plugin specification).