cogimon/rtt-gazebo-robot-sim #22: Problem creating force torque sensors

Pouya-moh, 2017-03-16:
I am at the head of this and gazebo-model repositories. When robot_gazebo is loading, we have four errors related to four sensors:
```
0.183 [ ERROR  ][ScriptingService] l_arm_ft_joint can not be associated to any FT sensor
0.183 [ Info   ][ScriptingService] Creating Force/Torque Sensor 
0.183 [ ERROR  ][ScriptingService] l_leg_ft_joint can not be associated to any FT sensor
0.183 [ Info   ][ScriptingService] Creating Force/Torque Sensor 
0.183 [ ERROR  ][ScriptingService] r_arm_ft_joint can not be associated to any FT sensor
0.183 [ Info   ][ScriptingService] Creating Force/Torque Sensor 
0.183 [ ERROR  ][ScriptingService] r_leg_ft_joint can not be associated to any FT sensor
```
It seems somehow `force_torque_sensor.cpp:21` returns null. [Maybe] related, is the fact that gazebo server is printing messages of `[Wrn] [msgs.cc:1655] Conversion of sensor type[force_torque] not suppported.` but according to gazebo [issue](https://bitbucket.org/osrf/srcsim/issues/68/gpu_ray-not-supported-with-nvidia-1050) tracker this is probably only a rendering issue. The force torque sensor outputs are indeed attached and can be read from ignition. Ideas?

Pouya-moh, 2017-03-21:
Updates:
I have located the problem, and it is strange... I have a ops file which its relevant part looks like this:
```
gazebo.spawn_model("coman", model_name, 10)
robot_gazebo.getModel("coman")

robot_gazebo.loadURDFAndSRDF(path_model_urdf, path_model_srdf)
robot_gazebo.configure()

gazebo.toggleDynamicsSimulation(true)
```
Here is the issue: If I run pass this to deployer as command line argument, in `robotsim::gazeboConfigureHook` xbot core finds all sensors from srdf, however, `SensorManager::Instance()->GetSensors();` finds zero:
```
std::map<std::string,int> ft_srdf = _xbotcore_model.get_ft_sensors();
// ft_srdf.size() is equal to 4 but
gazebo::sensors::Sensor_V sensors = gazebo::sensors::SensorManager::Instance()->GetSensors();
// sensors.size() is equal to 0
```
Here is the strange part: If I don't pass ops as arg and instead, run deployer and copy paste the stuff, then `ft_srdf.size()` is equal to 4 and `sensors.size()` is equal to 5! (4ft+1imu). So I guess the issue is not from urdf/srdf. Ideas?

xwavex, 2017-05-02:
@Pouya-moh can you please check if my fix also works for you? https://github.com/cogimon/rtt-gazebo-robot-sim/commit/fa064a975fae44cc7207abd002c8f8bcdf01756f I directly pushed it to the **master**, since it does not affect anything else.

**BTW** I do not get any IMU ports and I also do not find anything related to IMU in the source. Do you by chance have a different version than there is on Github?

xwavex, 2018-01-11:
Will close this, as it seems to work. Otherwise please reopen this issue!

commit fa064a97 (2017-05-02): solved timing issue related to loading of sensors. fixes #22
closed 2018-01-11 by xwavex