cogimon/rtt-gazebo-robot-sim #23: torque-control bug in resetKinematicChain

ndehio, 2017-05-17:
When calling the method "resetModelConfiguration" from the deployer for KukaLWR in torque-control mode, the control mode is automatically changed to position-control.

I guess the bug is in line 90 of file:
https://github.com/cogimon/rtt-gazebo-robot-sim/blob/master/src/kinematic_chain.cpp
The if-statement does not work as intended (no error message) and afterwards in line 94 the new control mode is set. Do you agree?

I don't understand yet that resetting a KinematicChain should only be possible for position-control mode... Could you please explain that?

##############################################
For you to reproduce the issue, here a minimum example:

import("rtt_gazebo_embedded") 
import("rtt-gazebo-robot-sim") 
require("os")
loadComponent("gazebo","RTTGazeboEmbedded")
setActivity("gazebo",0,10,ORO_SCHED_OTHER)
loadComponent("robot_gazebo","cogimon::robotSim")
setActivity("robot_gazebo",0,11,ORO_SCHED_OTHER)
gazebo.configure()
gazebo.start()
gazebo.spawn_model("robotmodel", "model://kuka-lwr-4plus", 10)
robot_gazebo.getModel("robotmodel")
var string path_model_urdf = os.getenv("GAZEBO_MODEL_PATH")+"/kuka-lwr-4plus/model.urdf"
var string path_model_srdf = os.getenv("GAZEBO_MODEL_PATH")+"/kuka-lwr-4plus/model.srdf"
robot_gazebo.loadURDFAndSRDF(path_model_urdf, path_model_srdf)
robot_gazebo.configure()
robot_gazebo.setControlMode("full_arm", "JointTorqueCtrl")
robot_gazebo.getControlMode("full_arm"); #JointTorqueCtrl
robot_gazebo.reset_model_configuration()
robot_gazebo.getControlMode("full_arm"); #BUG: JointPositionCtrl

ndehio, 2017-10-17:
Any ideas?

EnricoMingo, 2017-10-17:
This was a design decision since the robot is started in position Ctrl and when you reset the model it will restart as it was initialized the first time (so in position Ctrl).

Inviato da iPhone

ndehio, 2017-10-17:
ok thanks

closed 2017-10-17 by ndehio